MiSTer-devel/Hardware_MiSTer #6: problem with via tenting with SDRAM XSD v2.5

thioroup8, 2019-12-20:
Hi

I recently contact jlcpcb service to produce PCBs for SDRAM XSD v2.5. I used this gerber file: https://github.com/MiSTer-devel/Hardware_MiSTer/blob/master/releases/sdram_xsd_2.5.zip

Unfortunately, they sent me an answer containing question and I am unable to reply to them because of lack of knowledge.
Please find below a copy of their message.

> Hi Sir,
> 
> Well got your order with many thanks~
> 
> 
> Sorry to bother you, but there is an issue that we want to confirm with you before proceeding.
> 
> 
> As shown below, please kindly confirm that :
> 
> 1.the top side is Tenting via holes
> 
> 2.but the bottom side there is both "Enting Vias " and "oprning vias "
> 
> could ypu please kindky confirm that whether we can do your boards as your gerber file ??
> 
> 
> Your early reply will be highly appreciated!
> 
> 
> Thank you so much!
> 
[image]
[image]

Please help me to write appropriate answer to jlcpcb service.

Thank you for your attention.

sorgelig, 2019-12-20:
Time after time JLCPCB start to send stupid questions. This gerber file made many times by them without any questions.  
Here also quite stupid ask from them. It's obvious the vias on pads are opened while without pads they are tented. 2 columns of holes in the middles are holes for THT connector, not vias. 
Tell them to transfer approval to other more professional person.

thioroup8, 2019-12-20:
Thank you so much for your quick reply.
I already ordered all electronic parts from mouser and I was afraid to be unable to solder my own PCB due to JLCPCB manufacturer :(
I hope they understand my answer I will send to them following your instructions.
I understand it is not an issue in your design and so sorry to place this message in the wrong place...

thioroup8, 2019-12-20:
oups I clicked on wrong button.

thioroup8, 2020-07-13:
Hi

I am back again with my first try to solder a SDRAM board for my MiSTer.
Sadly I have some issues :( My memory board is not recognized by memtest utility.
So, I double check with my microscope and my multimeter. All seems correct.
So, I check another PCB without any component solder on it.
My multimeter detects unexpected connections: I mean, I am surprised to find thoses connections because I do not see them on gerber files. Please have a look on attached picture: unexpected connections in red.

What do you think about thoses connections? Is there any issue during PCB manufacturing?
[image]

sorgelig, 2020-07-13:
I don't see what unexpected there. GND to GND, VCC to VCC. Looks fine to me. Board has 4 layers, not 2.
I can't tell about manufacturing problem - it depends on specific manufacturer.

thioroup8, 2020-07-13:
On U2 chip, pin 54 is VSS
On U1 chip, pin 37 is CKE and pin 28 is VSS
So I do not understand "GND to GND"?
By simply looking at GTL, G1, G2 and GBL, it is difficult for me to deduce from these 4 gerber files the existence of those 3 connections (in red). I probably misunderstand gerber files: I need to improve my knowledge on this topic.

My SDRAM board is faulty and I do not know why:
- bad solders
- pcb issue
- component faulty
- ...

sorgelig, 2020-07-13:
VCC - is power, 3.3V
VSS - is GND. 
CKE is connected to VCC - correct.
Even on your pic you clearly can see VSS pins are connected to ground.
You can check connections using schematics.
I don't have more files than i upload.

thioroup8, 2020-07-13:
Well
I made a lot of checks on an empty PCB and I think PCBs are correct.
I suspect I used an excessive soldering temperature of 350 °C and I killed one or several components because datasheets recommend a temperature between 235 °C and 260 °C.
Stupid mistake to learn :(

david-cash, 2024-07-10:
I already ordered all electronic parts from [OVAGA](https://www.ovaga.com/) and I was afraid to be unable to solder my own PCB due to JLCPCB manufacturer :(