MiSTer-devel/Hardware_MiSTer #11: The MiSTer I/O board has serious issues with brightness levels when connected to a VGA monitor

mathieulh, 2020-03-06:
The MiSTer I/O board has serious issues with brightness levels when connected to a VGA monitor

When I plug a shilded coaxial VGA cable (ie, the best quality you can get) to the dsub 15 output from the IO board to my CRT VGA monitors (I tested 2 different ones) the brightness is very dim
almost as if you did not have adequate impedance or power output.

If I connect the exact same cable (and monitors) over direct video using the same MiSTer configuration the image looks great with no brightness issues whatsoever.

This appears to be hardware issue related to the I/O board itself.

sorgelig, 2020-03-07:
It must be very specific to your setup/monitor. 
Usually VGA output provides good video. I have tried many monitors - no problem with brightness. Anyway, if more strong signal is needed then it will require additional amplifiers. It won't happen in I/O board as there is no place on board and 5V power will not be enough for analog amplifiers. 
Use Direct Video - it provides more bits per color and better audio DAC.

mathieulh, 2020-03-07:
Hum... I am having the issue on both my CRT monitors, a Sony Trinitron CPD-200GST and a Compaq P1220 (Mitsubishi Diamondpro 2060U rebrand), the only thing I have not tried yet is swapping the I/O board as I do not own another at this point. I also tried different cables.

sorgelig, 2020-03-07:
May be wrong video DAC resistors are used on I/O board.

twinkytwonka, 2020-06-14:
> Hum... I am having the issue on both my CRT monitors, a Sony Trinitron CPD-200GST and a Compaq P1220 (Mitsubishi Diamondpro 2060U rebrand), the only thing I have not tried yet is swapping the I/O board as I do not own another at this point. I also tried different cables.

If it's a V6 or 6.1 io board then what resistor array pairs did you use? Your choice has a significant effect on brightness levels . If you're using a 5.7 board then double check components with the schematic

closed 2022-11-27 by sorgelig