MiSTer-devel/Hardware_MiSTer #12: Pulsing LEDs on powerup, nothing else happening!

Galbi3000, 2020-03-06:
I've not used my MiSTer for a few months (been busy PC gaming instead of retro). It's just been sitting on my desk in all the time unpowered with the same SD card in since the last time I used it, not taken it out to update or make any changes.
Today I decided to boot it up again as I got some new ROMS I wanted to try out, I switched the MiSTer on and nothing happens. Well, nothing that should happen.

What I am getting instead of the MiSTer bootup sequence on my TV is no signal and the addon board pulsing the red and yellow LEDs (labled POWER and USER respectively). The smaller red LED is just on.
Under the addon IO board I see the green LEDs beside the ethernet port also pulsing the first and fifth LEDs.

What is going on? Like I said, the last time I booted it up it worked fine and it's just been sitting there unpowered until today.

Here is a GIF to show what the LEDs are doing:

[image]

sorgelig, 2020-03-06:
Please use forum for discussion. Github is for specific issues.

Galbi3000, 2020-03-07:
How is this not an issue? The thing is not working and I do not know why!

sorgelig, 2020-03-07:
LEDs flashing is correct.
post your problem here:  http://www.atari-forum.com/viewtopic.php?f=117&t=34710

closed 2020-03-07 by sorgelig

Galbi3000, 2020-03-07:
Well, if the LEDs flashing is correct, that still leaves the fact I have a black screen on my TV, there is still an issue of the device not functioning as it should! It is in my description of what is wrong and in the issue title "nothing else happening!". You are always so quick to just dismiss an issue it seems. I have started actively pointing this out on YouTube videos talking about the MiSTer, an otherwise good product let down by poor support. Good bye.

sorgelig, 2020-03-08:
Ask those where you've bought the product. I didn't sell you anything. 
This is my hobby and i do this for free.
Please leave and don't come back again!

RetroShopPT, 2020-03-08:
> Well, if the LEDs flashing is correct, that still leaves the fact I have a black screen on my TV, there is still an issue of the device not functioning as it should! It is in my description of what [...]

Hi, look like a bad sdcard update and not hardware issue. Anyway, from photo, look like a board sold by me, and if you have any problem with hardware, just send me a message on Facebook or on my site.