MiSTer-devel/Hardware_MiSTer #15: Request: STEP or STL files for each board for easier CAD design of accessories.

birdybro, 2021-04-14:
I would love if I could load a STEP or STL file of the boards and their components in one of the many free CAD programs out there to begin designing 3d printed cases and accessories. Unfortunately this currently requires extremely expensive premium software to even export these, and there are tools online like altium2kicad but they are not very functional. I realize that moving to open source PCB design tools is undesirable however can some of the paywall-locked features like STEP/STL export from the Altium format be at least provided for us here?

Thank you!

KhalidHisham, 2021-04-19:
you mean like this? 
[sdram_xsd 3D-model.zip](https://github.com/MiSTer-devel/Hardware_MiSTer/files/6337187/sdram_xsd.3D-model.zip)

I am new to gethub so I am not sure how to upload the files to the repo otherwise I can help. 

\K

KhalidHisham, 2021-04-20:
Thanks, I will do it as soon as I can. 
Perhaps it will be cleaner to put all the step files in one folder?

birdybro, 2021-04-20:
Maybe. I usually try to match the folder structure that the repository
already has to make it more agreeable to the maintainer of that repo.

KhalidHisham, 2021-04-20:
Ok so I generated everything but it seems i can't add the files using GitHub desktop. for some reason no changes are shown when I simply put the files in their respective places. Is there a way to fix?

birdybro, 2021-04-21:
Looks like it worked!

[image]

Next you have to submit the pull request.

[image]

[image]

[image]

:)

KhalidHisham, 2021-04-21:
That was quite detailed, 
Thanks :)

birdybro, 2021-04-21:
Glad to help. Welcome to GitHub! 👍🏻

KhalidHisham, 2021-04-22:
Thank you. It is done now. 

I found out the problem I had with the gitHub desktop. It seems that the all STEP files are placed on ignore list in the repository configuration. So I decided to zip all the files before adding them to their respective folders and it worked.

sorgelig, 2021-04-23:
it's on repo now

closed 2021-04-23 by sorgelig