MiSTer-devel/Hardware_MiSTer #22: Help with usb board

sonik-br, 2021-12-27:
Hi,

My USB board is busted. The component over the U2a (NCP380LSNAJAAT1G) is burned and I can't find another to buy.
There's any equivalent component that will work?

[image]

sorgelig, 2021-12-28:
option #1: you may disable VBUS control, so you won't need this chip. USB devices will have permanent 5V supply from main power. Refer to schematics and notes there.
option #2: there are no equivalents. There is only version for uDFN6 case of the same chip which may hold slightly higher current than busted TSOP6 version.

sonik-br, 2021-12-28:
Thanks for the answer.

I'm a layman in electronics .

What's the purpose of VBUS? If I get it right, it's to power on/off the usb hub in sync with mister's main board?

For option 1 I can just bridge the JP3? JP1 too?
It will drain power from the mister's usb or from external psu?

sorgelig, 2021-12-29:
> What's the purpose of VBUS? If I get it right, it's to power on/off the usb hub in sync with mister's main board?

it allows USB host controller to turn power on/off of USB devices.

> For option 1 I can just bridge the JP3? JP1 too?

Short JP1. JP3 doesn't matter.

> It will drain power from the mister's usb or from external psu?

It will drain power from the same source as with VBUS control.

sonik-br, 2022-02-05:
My board is dead. Even without VBUS. Power works but no data.
I've got a new one :)

closed 2022-02-05 by sonik-br