MiSTer-devel/Hardware_MiSTer #23: Solderless MT32Pi_Lite using pogo pins

tmznix, 2022-02-03:
Is it possible to modify this board to use pogo pins instead of soldering ?

Something like this: (examples)

https://thepihut.com/products/pogo-a-go-go-solderless-gpio-pogo-pins

Or this:
https://pinout.xyz/pinout/zero4u#

And the last one ( I promise :) )
https://www.kickstarter.com/projects/pico-learber-board/notixed-turn-your-pi-zero-to-usb-device

Cheers!

sorgelig, 2022-02-04:
From PCB point of view, no modifications are need. Since pins are too high, you will need to place MT32Pi_Lite under the RPi so USB3.0 plug will be able to fully inserted. Overall construction will be very thick and bulky. If you like this then use it. Original idea was to eliminate connectors and make construction as slim as possible.

rataweb, 2022-03-02:
Hi,
It's a good idea.
Please, Could you upload Altium pcb files in ASCII format? I'm using Eagle and it's the only way to import Altium files. I'll try to modify by myself.

Many thanks
Shark