MiSTer-devel/Hardware_MiSTer #26: [Feature Request] Hardware_MiSTer/io_brd_6.2 with dedicated DAC circuit

ArcadeTV, 2022-11-27:
I use io_brd_6.1 on my MiSTer Setup with [this external DAC](https://amzn.eu/d/gATQUuJ) for better audio. 

[image]

To minimize the need for cables and power supplies I had the idea to merge the DAC's circuit into the io board. Before I start creating a pcb for modding io_brd_6.1 I wanted to ask if this may eventually be added to io_brd_6.2. By merging the circuit into the io brd R6 (75R), the AMS1117 regultor and its components and the coax-parts become obsolete. [Here](https://www.dropbox.com/scl/fo/4myo1c3tk06i6u13dm5p3/h?dl=0&rlkey=h2cjf12vrrlcghgtq57sm88s8) is my Dropbox folder containing my scans and datasheets.

I don't know or have Altium, so I'd like to know if it would be possible to export the io_brd_6.1 to eagle or if the DAC circuit could be included in the next version of the pcb, maybe with an addional switch to toggle between native audio and DAC output.

If this may be rejected, I plan to create an addon-pcb that may be positioned here and using the existing pads:

[image]

---

For prerequisites I stripped the DAC and drew its schematic:

[image]

[image]

sorgelig, 2022-11-27:
I suggest to make an addon board using I2S header from I/O Board, so it will follow modular structure of MiSTer. I use it as well for some my embedded MiSTers. It will eliminate CS8416 as well. Just need to find I2S DAC which can work without MCLK. Current I/O Board has no free space for such circuit, so embedding it to board will require a denser routing and smaller components which i don't want to do. I want to make it DIY friendly. 

If you will do an addon board then may be it would be better to make a tiny board which will plug directly to that i2s connector and use 3.5mm jack on top of existing one. Just thoughts..

Thinking more about this: most modern high-end amplifiers have either optical or coaxial input which is anyway better than analog. Your board is mostly for integrating into some arcade cabinet where such amplifier is not available. Thus using original external converter should be fine already. If your amplifier has no digital input, then you may place this converter near your amplifier and hide somewhere there, so for MiSTer you will need only thin optical cable.
HDMI btw, has no such issue as it delivers digital audio. In direct video mode you can use HDMI-VGA converter with audio output.
So, basically there are many options for better audio are available.

sorgelig, 2022-11-27:
btw, i even not sure if integrated converter on I/O board will have enough noise over power/gnd isolation. Analog part should be well isolated which separate board with separate power supply does.

sorgelig, 2022-11-27:
not sure your original intention of using this converter.
In case if you use HDMI output and simply want to connect some small stereo speakers, then this converter probably will be more suitable:
https://www.amazon.com/Optical-ROOFULL-Extractor-Splitter-Converter/dp/B01LL5RI3Y

ArcadeTV, 2022-11-27:
Thank you very much for taking the time to answer, this is very much appreciated.
The motivation for this is to NOT have extra external hardware and power supply. The device works great, but I'd love to have it inside the MiSTer. I'm not a professional, so any hint about best-practise and isolation is great. Maybe additional coils or EMI-filters could cut the noise?

This is what I had in mind, piggyback an add-on-pcb under the I2S and VGA-PWR headers, so they can be installed back and stay functional. For L and R audio-lines, either the traces could be cut or C11 and C12 could be removed and DAC-output connected to their pads.

[image]

This is a mockup of what it could look like, all the parts are already present.
This is not as modular as I'd love to make it, but if it works it would be beneficial for me.
Since I use CRT's only, I never connect HDMI.

retrorepair, 2022-11-28:
It seems to me all you would need there is a pin header where you have the pads currently and maybe a dip switch/jumper block to select built in audio and I2S on the jack output. The wouldn't impact the IO board's design at all and then everyone's happy?

It has to be said, using those cheap Amazon DACs gives a vastly superior analog audio output than what's built in. I'm not much of an audiophile but it's a huge step up. Eradicates the pops on core change too.

ArcadeTV, 2022-11-28:
Indeed, that would solve the problem. I did some routing in the meantime.

[image]
[image]
[image]

retrorepair, 2022-11-28:
Looking good.

Rather than cut the board traces, you could always remove the two large caps under the IO and hijack the audio output from the two vias leading to them. I think your board would need to extend to the far one though and I believe your caps/resistor may currently be in the way of the other.

I always find soldering to cut traces a bit hacky and often leads to poor connections.

The other alternative would be to cut the two traces leading to the I2C header before then hit RN17 and the top side traces to the GPIO, then either run jumper wires from the I2C to the audio jack or maybe even a nice QSB to keep it somewhat tidy? Hijacking those now spare pins of course with your own audio outputs.

ArcadeTV, 2022-11-28:
> Rather than cut the board traces, you could always remove the two large caps under the IO and hijack the audio
Yes, I already wrote that C11 and C12 could be used for that.

> I think your board would need to extend to the far one though and I believe your caps/resistor may currently be in the way of the other.
I'm afraid there's zero space left to move anything around, so maybe the wires need to go from top to bottom - which would be ok when using a shileded cable

> You plan to add it as open source in the MiSTer GitHub?
I didn't plan ahead tbh, but I will create a repo with full source files once it's tested. I'm afraid, as sorgelig pointed out, there may be issues with ground loops and noise when run off the internal MiSTer power, so maybe coils and emi filters have to be added, but I have no experience with that.

n67t1cq, 2022-11-29:
It is good to implement PCM51002 into IOBOARD.

I've made i2s addon board for 6.1 last year. It works good.
[image]
[image]

ArcadeTV, 2022-11-29:
> PCM5102a

That's really cool and I wish I had know this existested before I started ;-)
Where can I get one? I've read that this IC is programmable - would you share some insights how it works?

ArcadeTV, 2022-12-01:
@n67t1cq could you share some insights?
Is it for sale or maybe even open source?