MiSTer-devel/Hardware_MiSTer #28: MT32Pi Top Gerbers broken

Shadowtrance, 2023-08-02:
Title.
It seems the gerbers for the MT32Pi Top piece are broken in some way.
When imported to jlcpcb this is the result (1).
And if you select the top or bottom after initial loading (2) you only see a corner of the pcb with a hole that isn't drilled out and not much else (3).
The main base board that actually matters _appears_ to be fine though.

1. [image]

2. [image]

3. [image]

sorgelig, 2023-08-02:
Their online gerber viewer is very buggy. Don't verify correctness there.

Shadowtrance, 2023-08-02:
Well it wasn't just jlcpcb. I checked it in kicad gerber viewer and some others online all with slightly different but weird results.
I thought I was going a little crazy... hopefully not. haha Main appears to be fine on JLC in 2d and 3d views, but then different in kicad gerber viewer. So I don't know what to believe...

Top [image]

Main [image]

sorgelig, 2023-08-02:
It's relative/absolute coordinate mistreat between drill and gerber. Don't worry. JLCPCB stuff will correct it. All boards i release I've ordered from JLCPCB already.

Shadowtrance, 2023-08-16:
Ok yeah i have no idea what weirdness was happening with every gerber preview i looked at but i ordered them anyway and they can in today perfectly fine and not some weird cutout like the jlc preview showed for the top.
All good and thanks for creating the addons! :)

closed 2023-08-16 by Shadowtrance