MiSTer-devel/Hardware_MiSTer #30: fix suggestion: remove possibility of wrong pin insert alignment

goofyseeker311, 2024-04-15:
if there would be a socket, that would fit with the de10-nano in only one way,
or that there would be a protection circuit in case of wrong pins, there would be
no way to brick any of the boards. just saying as feedback as an user.
if you can break it, you will break it, like with the sdram board pin misalignment, into wrong pins.

sorgelig, 2024-04-15:
If de10-nano would use shrouded connector like IDE HDD, then it would exclude wrong insertion. But DE10-nano uses naked pin header, so there is no way to prevent it. Adding protection circuit is impossible too.

When you buy a TV or phone, you buy it fully assembled. So do it with MiSTer too. Leave DIY option to professional people.

goofyseeker311, 2024-04-15:
I'm a professional electrical engineer, and of course all is possible. there is a way to make a suitable socket.
but you do you.

closed 2024-04-15 by goofyseeker311

sorgelig, 2024-04-16:
I don't sell anything, don't you know? I don't even get any revenue from sales besides some sellers donate sometimes to me.
This is hobby and community project.
Generally speaking it's supposed that you buy all components and solder all addon boards yourself bearing all risks on yourself.
If you buy a Raspberry Pi or Arduino boards and then randomly connect the pins, then any damage you made by that is not covered by board manufacturer. Same with MiSTer. 

As a community/hobby project it's developing with help from people wanting to share their improvements. Since you are professional (as you said), you may offer protection circuit or other improvements preventing wrong insertion.

goofyseeker311, 2024-04-16:
just bringing up potential fixes/improvements, if someone wants to incorporate them.

sorgelig, 2024-04-16:
It's easy to suggest and it costs nothing. Just make a prototype which suits everything you've said. Spend the time, money, nerves and so on. 
Then come here to share everything here for free. 
Agree?

goofyseeker311, 2024-04-16:
only half, its better to first state what the improvement is, so that anyone can do it, even if I would not be doing it.
then even I can do it, better have a good plan first, half the work, agree?

goofyseeker311, 2024-04-16:
about the being a "pro", I do hold an uni master/semi-phd in electronics/semiconductor/nano-tech.

goofyseeker311, 2024-04-16:
I dont like to be forced either.

this is how it works in the corporate world, where I would be a lead designer, not the implementer.
mostly being on the design requirements side of product process. but I could/can do it myself when I want to.

sorgelig, 2024-04-16:
And there is 5V pin on connector, and whatever signal it collided on wrong insertion, it will damage the board because SDRAM is 3.3V. Go to protect it ;)

goofyseeker311, 2024-04-16:
yes that 5V vs 3.3V is an issue, but it can be dealt with.

sorgelig, 2024-04-16:
> I would be a lead designer, not the implementer.

I see, lol :)

goofyseeker311, 2024-04-16:
I say this before even looking at the circuit design.
so how about the series resistor on all the data lines.

sorgelig, 2024-04-16:
> so how about the series resistor on all the data lines.

No problem. You can try it and report it here.

> or maybe just a voltage divider from 5v to 3.3v.
already added in recent SDRAM revisions. It has nothing to do with protection. Shifted insertion will put 5V on nearby signal.

goofyseeker311, 2024-04-16:
it has all to do with the data line protection, to prevent excessive current and voltage. its about mass current at 5v to actually burn stuff on wrong pin, not the 5v signal alone. it would not work with the 5v signal would it would not burn.

goofyseeker311, 2024-04-16:
yeah I would just put a resistive voltage divider on the data lines, even if it has high output impedance not matching the sdram chip impedance.

sorgelig, 2024-04-16:
couple more hours and you will say that current board design is fine.
Laziness is the strongest force!

goofyseeker311, 2024-04-16:
lol :)

goofyseeker311, 2024-04-16:
this is fine. lol.