MiSTer-devel/Hardware_MiSTer #32: [Feature Request] Saturn 10-pin mini DIN AV port for the next official I/O board

thericeking, 2024-11-25:
Has any consideration been given to adding a Saturn 10-pin mini DIN AV port to the next official I/O board, similar to the MiSTerAddons (Porkshop Express) IO Direct board? Forgive my ignorance, but it looks like it would allow the option of dual RAM analog operation and HDMI, albeit not at the same time. It is still really nice having both options. I'm very curious to know what's your opinion on this. I apologize in advance if this has been requested and answered before, but I couldn't find any discussion about this on the Internet. Thank you in advance in taking the time to respond. Take care.
 
[image]

sorgelig, 2024-11-25:
It's just a connector. It won't give you dual-SDRAM and Analog output at the same time. MiSTer Addons I/O board has integrated HDMI->Analog converter - that's how it provides analog output. You can achieve the same by using external HDMI->Analog converter.
Saturn connector is less used than VGA connector and requires quite specific cable. MiSTer goal is to use as popular as possible connectors for easy connection. That's why I/O board uses VGA connector.

thericeking, 2024-12-09:
Thank you for taking the time to respond.

closed 2024-12-10 by sorgelig