MiSTer-devel/Hardware_MiSTer #33: Analog Audio Out Distortion

TheJesusFish, 2025-06-26:
TL;DR: The stock designed 9.2 board (with io mezzanine) has an issue with audio out in some cores and some games. I haven't done a full barrage of testing, but it's very easy to hear in Keio Flying Squadron for the Mega CD. Further, it seems like this may be related to the vga_mode setting in the MiSTer.ini.

As referenced [here](https://github.com/MiSTer-devel/MegaCD_MiSTer/issues/112#issuecomment-3002300641) and (here)[https://github.com/MiSTer-devel/MegaCD_MiSTer/issues/114], there has been some weirdness with analog audio out on the 9.2 boards (we see folks bringing it up in teh discord help channel as well).

I decided to buy a bunch of boards to test and this issue is **not** limited to the Mister Pi 9.2 board. I would imagine that is just the most prevalent IO board based on the stock 9.2 design, and that's why folks having this issue are more likely to have it. I have included the recordings in a zip.

Testbed:
- de10 nano
- MiSTer Addons 128mb 3.0 SDRam
- Stable Mister and cores (all updated)
- Games stored on the SD card.
- Global Volume set to 5 bars

Keio Flying Squadron tests:
- IO 6.1 board and vga_mode=rgb (control): No distortion
- MiSTer Addons Analog Pro vga_mode=rgb (9.2 board - non-stock design): No distortion
- Retro Remake 9.2 Pro vga_mode=rgb (stock design): heavy distortion
- Legacy Pixels 9.2 Pro vga_mode=rgb (stock design): heavy distortion

Further more:
- Legacy Pixels 9.2 Pro vga_mode=svideo (stock design): light (no?) distortion
- Legacy Pixels 9.2 Pro vga_mode=ypbpr (stock design): no distortion

I'm happy to do further tests with other games that have reported problems, but this felt like a good jumping off point if you had any appetite to look at this.

[Sound Recordings.zip](https://github.com/user-attachments/files/20917774/Sound.Recordings.zip)

bootsector, 2025-06-26:
Taki boards are created as per specs and are very high quality as far as I can tell. My Taki 9.2 analog board has the same issues but they seem to get "fixed" if I play around with the vga_mode settings or use one of the "shifted" cores. I think that if those were related to hardware//quality issues then the issues would persist regardless? 

Oh, an optical DAC solves the issue as well, so that's what I'm using for now.

TheJesusFish, 2025-06-26:
Luckily, the issue does not affect a lot of games. Based on what I have seen in the discord, it is almost entirely isolated to the Mega CD and PSX cores. And even then, a very limited amount of games.

As Boot mentioned, Robert built a shifted PSX core troubleshooting something else and that core happened to fix the sound distortion issue for the folks that were experiencing it. That build changed nothing audio wise, so Robert wasn’t sure why it suddenly fixed the issue (especially since timings were met during the build).

I’d be more than happy to send you both reference-based boards if you want to compare their construction to your designs. I didn’t mean to insinuate it’s a hardware issue. I only filed it here because it was related to the boards and not a singular core/game.

TheJesusFish, 2025-06-29:
Same everything (de10, power source, sd card, and wired controller). Only thing that changed was the IO Board.

sorgelig, 2025-07-01:
I have no distortion on my IO 9.2. Tested on USA version of Keio game.
So, it's weird why some boards have such distortion and only specific game... I have no idea, unfortunately.

paulb-nl, 2025-07-01:
Thanks for testing @sorgelig . You have `vga_mode=rgb` in ini?

TheJesusFish, 2025-07-01:
So strange.

I have tried those two boards in a couple of different DE10s and Mister Pi boards - all the same result (that’s why I finally made the issue because it felt like a smoking gun). I wonder if there is a way that folks are manufacturing these that differs from the boards you built. I haven’t tried the 9.2 board from UltimateMister or Misterfpga.co.uk yet - going to see if someone on the discord has them.

The recording I included above is with global volume at 5. I didn’t touch core volume, so that’s potentially another thing to check.

But, either way, the problem disappears when using a different seed to build both cores. I dunno if that means that there is an issue with the way both cores are written, or if the build is assembled down a certain path it triggers a hardware issue. Either way, it can be mitigated by changing the seed (everyone’s favorite solution :) ).

I thought for sure we had finally found a way to track this one down, though.

nekrondev, 2025-07-01:
For the context: my Mister Pi (turbo pack, batch 2 from December 2024) doesn't output the distorted analog audio TJF posted here. It seems to be random silicone lottery...

TheJesusFish, 2025-07-01:
> For the context: my Mister Pi (turbo pack, batch 2 from December 2024) doesn't output the distorted analog audio TJF posted here. It seems to be random silicone lottery...
> 
> Edit: Wait, let me crosscheck if `vga_mode=rgb` is set (I didn't modify Takis default presets)... ok, not found in the `Mister.ini` but docs tell `vga_mode=rgb - Changes the VGA output to a rgb, ypbpr, svideo, cvbs. rgb is default.` I additionally added it and rebooted the device, but audio seems to be fine on my headphones.

I wouldn't use Taki's mister.ini - not sure what they have changed. Grab this one: https://github.com/MiSTer-devel/Main_MiSTer/blob/master/MiSTer.ini and test again if you have a chance.

AtelierBleu, 2025-07-01:
I’ve attached uncompressed, 24-bit 44.1khz stereo audio recordings of the Keio intro with vga_mode=svideo (clean) and vga_mode=rgb (dirty). I also noticed that the loud audio pops appear in the left and right channels at different times rather than simultaneously.

[image]

[keio-audio.zip](https://github.com/user-attachments/files/21008737/keio-audio.zip)

birdybro, 2025-07-02:
@RetroShopPT Thank you for offering to look into this. Good luck with the busy schedule.

When you do check it out, can you confirm that the PLL to the PCM5102 DAC is stable/reliable? Switching the OSD on and off, turning on and off composite blend, changing these vga mode options, etc... This stuff might make the output pll unstable to the DAC maybe? Or maybe some kind of coupling from the video path is making it's way out the output ports alongside the audio. I'm not sure, just throwing some stuff out there. It really seems like some kind of interference from the digital domain on the board is making it's way up the gpio.

nekrondev, 2025-07-02:
> I wouldn't use Taki's mister.ini - not sure what they have changed. Grab this one: https://github.com/MiSTer-devel/Main_MiSTer/blob/master/MiSTer.ini and test again if you have a chance.

For the record: I added the devel mister.ini as an alternative config setting and tested again with Keio (US, CHD from SD card), but found no crackling noise.

birdybro, 2025-07-02:
@sorgelig two users now said that the test build I did with an updated framework solved the issue. @dshadoff suggested this test.

https://github.com/MiSTer-devel/MegaCD_MiSTer/compare/master...birdybro:updatesys

Could any of the following be responsible?

[image]

AtelierBleu, 2025-07-02:
> > I wouldn't use Taki's mister.ini - not sure what they have changed. Grab this one: https://github.com/MiSTer-devel/Main_MiSTer/blob/master/MiSTer.ini and test again if you have a chance.
> 
> For the record: I added the devel mister.ini as an alternative config setting and tested again with Keio (US, CHD from SD card), but found no crackling noise.
> 
> Edit: Please note that I'm outputting video through HDMI and only connected by headphones to the analog audio out. Not sure if load on the VGA connector has something to do with it.

I do get crackly audio even without anything attached to the VGA port.

TheJesusFish, 2025-07-03:
Best path forward is to wait for Ricardo and then (assuming his testing comes back normal) and then close this issue? I know the discord will love that SRG has another core to work on instead of 3DO :D

TheJesusFish, 2025-08-13:
Closing this as it seems to be metastability. MegaCD unstable doesn't show the issue and the released PSX core is also fine.

RetroShopPT, 2025-08-13:
I’ll only be able to finish the tests in about two weeks. Unfortunately, a family member passed away, so I’m running behind on everything, and then we have the holiday break. Still, I’ll be available to help in any way I can and run tests with several boards. Best regards to everyone.

Ricardo

psoma, 2025-09-05:
Only bought a CRT recently (was using digital/HDMI prior) and noticed this issue on Sonic CD, particularly during the intro which led me here.

Have a taki Mister Pi with 9.2 IO and outputting ypbpr.

Using https://github.com/birdybro/MegaCD_MiSTer/raw/refs/heads/updatesys/releases/MegaCD_20250701.rbf reduces the clicking/popping during the FMV portion of the intro, but it is still definitely present.

Experienced the same with the PSX core but was pointed to an updated rbf that completely addressed the issue.

To reiterate, the 20250701 build improves the issue, but certainly doesn’t eliminate it fully.

TheJesusFish, 2025-09-06:
@psoma Please try the unstable: https://github.com/MiSTer-unstable-nightlies/MegaCD_MiSTer/releases/download/unstable-builds/MegaCD_unstable_20250703_077d86.rbf

jotego, 2025-10-06:
Was the culprit for this identified? Is it some timing issue in the core compilation or is it a hardware problem on the board itself?

TheJesusFish, 2025-10-06:
> Was the culprit for this identified? Is it some timing issue in the core compilation or is it a hardware problem on the board itself?

It was not. @RetroShopPT was going to take a look but real life has not been kind to them so I don't think they have had a chance to look.

TheJesusFish, 2025-10-21:
The current bubble bobble core still suffers from the issue (which would make sense if it was unchanged and the seed was not altered).

alberthorta, 2026-11-01:
Another case, with the SNES core.

- Hardware: QMTECH Cyclone V board + QMTECH I/O board (connected through a custom backplane, not stacked). Analog audio from the 3.5 mm jack of the I/O board.
- Core: `SNES_20260823.rbf`
- Game: Super Mario World (USA)
- Screen: the overworld map, when Mario walks on the map. The level screens are fine.
- Nothing connected to the VGA port.

Results:
- `vga_mode=rgb`: the analog audio is distorted on the map screen.
- `vga_mode=rgb` + `composite_sync=1`: still distorted.
- `vga_mode=ypbpr`: clean.
- `vga_mode=svideo`: clean.
- HDMI audio: always clean.

For now I use `vga_mode=svideo` as a workaround.